fra589/cn5X #1: Language Support

liushuya7, 2018-05-01:
Hi @fra589,

This program looks really awesome! But I can't read French. Is it possible to switch to English interface?

Thanks,
Joshua

luisluna2, 2018-05-04:
@fra589 I can traslate to Spanish!

fra589, 2018-05-04:
You are welcome @luisluna2,  
But for now, the multilanguage support is not implemented... Be patient !-)
@++;
Gauthier.

luisluna2, 2018-05-04:
@fra589 Okay, waiting for... :1st_place_medal:

jflapao, 2018-05-05:
@fra589 I help you translate to Portuguese form Portugal, and preaps Portuguese from Brasil, I can Speak and write in Spanish, so I can help you in this language to.

fra589, 2018-05-24:
HI @all,
I published today the first release with language support (actually French & English).
Click on the flag at the right of menu bar to change.  
Language support is made with native QT5 i18n support (http://doc.qt.io/qt-5/internationalization.html).
If you stay wanting to help, you can download i18n/cn5X.*.ts and use Qt Linguist (or any other editor) to make translations and send me the new file.  
Thanks in advance for your participation,
@ += 1
Gauthier.

closed 2018-07-13 by fra589