fra589/cn5X #2: Jogging

liushuya7, 2018-05-02:
Hi @fra589,

Is there a way to adjust the jogging speed in cn5x?
[image]
screenshot from Laserweb

Thanks,
Joshua

fra589, 2018-05-03:
Hi @liushuya7,

This functionality is not yet developped. Jogging actualy use one speed for all moves, this speed is defined in line 35 of cn5X_config.py

@++;
Gauthier.

liushuya7, 2018-05-03:
Got it. Thanks! Look forward to having it being added on.

fra589, 2018-05-24:
Functionality added to release 0.3 published today :  
https://github.com/fra589/cn5X/releases/tag/v0.3-alpha
I hope this help you,
@ += 1
Gauthier.

liushuya7, 2018-05-24:
Great! Thanks!

closed 2018-05-25 by fra589

cneve, 2018-09-22:
Can the 0.3 version run on Debian?