fra589/cn5X #4: Support for non-GRBL too?

justinclift, 2018-07-06:
This looks like an interesting new 5-axis software control project. :smile:

Is the focus on just supporting 5-axis GRBL, or should it (eventually) support other 5-axis controllers too?

eg [g2core](https://github.com/synthetos/g2), [LinuxCNC](https://github.com/LinuxCNC/linuxcnc), [MachineKit](https://github.com/machinekit/machinekit), etc?

justinclift, 2018-07-09:
Cool. :smile:

Yeah, I saw this repo after coming across your 5/6 axis version of Grbl.  I've been using a Shapeoko 3 (uses GRBL) for um... probably about 2 years now.  But, when looking for a way to add further axis, things turned out to be more challenging than expected.

I had come across a 4-axis fork of GRBL, but as my personal goal is to make a 5-axis CNC machine (about Shapeoko 3 size in X/Y dimensions), the lack of more axes seemed to write it off.

Instead I went with g2core (the successor to TinyG), as that has the needed support already + lots of enhancements (proper motion control!) & various CAD packages output 3-axis G-Code for it already.
( In theory, it *shouldn't* be too hard to expand that G-Code to 5-axis output, but we'll see ).

As a thought with your project, are you thinking of upgrading the motion control aspect to include stuff like jerk control?  If you are, maybe take a look through the g2core code, as the licenses look to be compatible? :smile:

fra589, 2018-07-12:
Hi Justin,
I think I haven't enough mathematical skills to include jerk control in the motion control of grbl-Mega-5X.  
But supporting g2core firmware in the cn5X contrôl panel is an interesting option for later...  
@++;  
Gauthier.

closed 2018-07-12 by fra589