fra589/cn5X #5: Some Issues some suggestions

jflapao, 2018-07-11:
I'm start using your cn5X program in my mpcnc, I know is in alpha version, but after read some post on the forum I realise that is only my endstops that not work.
My CNC is based on a mpcnc of vicius, and we use the marlin firmware to run the CNC that is, 3D printer to, in the first stage of my CNC using the mega and the marlin mpcnc modified firmware, all my endstops work the x+,x-,y+,y-,z+,z-, the machine still the same, and have the same wires, but because i don't like the repetier host, or printrun to run the milling jobs (is more like to 3d printing), i'm start using a arduino uno with the 0.9j version of the GRBL, and i don't know why my endstops stop working, i have work arround with the $20 $21 and $22 and nothig so i have give up, now with this version of the GRBL to the mega with 5/6 axis that i'm plan to implement i'm start trying again but still the same if i put the mega with the marlin the endstops works, the mega with the GRBL the endstops don't work. so if you can help me i appreciate.

The suggestion is, in your cn5X program you have to bottoms for the coolant, well i can't see the difference between the food and the mist, for me if you need the coolant and i work with a big GMM CNC for stone we use flood because we need a lots of water to cool down the tool and the stone. For us maybe is more useful have only the coolant flood for stone or metal, and a vacuum (M10 and M11 g-code command) for wood, or even better in the options menus have a way to personalize that. Other thing that happened to me have respect to the laser mode in the configuration, I have laser mode on, because when i work with the spindle use the M3 and M5 command to activate a relay that turn on and off my spindle and the M4 and M5 to operate the laser, but in your programe if the laser mode is on the relay stay on and off when the z axis goes up and down, so still good ideia put a bottom to activate the laser mode outside the configuration.

fra589, 2018-07-12:
Hi @jflapao,  
  
On my testing platform, the end stops are working fine, so I need more hardware info for helping you...  
Do you use specific hardware for your end stops ? Are they normaly open or normaly closed switchs ? Does they are all on the same type ?  
Do you try to use normals switchs connected between end-stop pin and GND ?  
  
The goal of my cn5X project is to make an interface for Grbl wich support up to 6 axis, implementing all the Grbl functionalities, without need of changing anything in the input GCode. All the computations must be made by Grbl, cn5X just interpret it's status result for displaying it.  
  
In Grbl, there is 2 coolant mode : Flood and Mist. On many professionals CNC, those 2 mode are for light watering like rain (Mist) or strong watering (Flood). The M8 command (flood) send tension on port D10 of Ramps card and the M7 command (mist) send tension on port D9 of Ramps card. M10 and M11 are not supported by Grbl, neither by Linux CNC which is the base model of it.
For your need, perhaps can you connect your vacuum on the D9 Ramps's port and use M7 for activate it.  
  
In the GCode standard, M3 is for starting the spindle clockwise, M4 for starting counterclockwise and M5 for stop the spindle. Thoses command are send to Grbd without any change. The laser mode of Grbl is not known by cn5X, it is only a parameter of Grbl wich change it's behavior. Dynamically changing the Grbl parameters can be hazardous and I don't want to do ti. That's why you need to have Urgent stop active to acces to the parameter panel.  
  
@++;
Gauthier.

fra589, 2018-07-13:
@jflapao  
  
I don't know why your end stop don't work under Grbl... I have exactly the sames and they work whell for me... The only advice that I can give you is to check the addresses of the ports in the file cpu_maps.h  
You can also verify the end-stops status leds at the left bottom of cn5X windows, perhaps they work well, but with wrong configuration in Grbl ($21, $22 & $23) tey don't stop the movement.
  
About the M10 M11 commands, it's sure it's possible to add it in my version, but it will take time to developp, then time to maintain, all for a non standard functionality wich will be used by a very small number of users... I prefer to focus my attention on standard elements used by as many people as possible.  
  
To toggle simply the laser mode, you can use the GCode input : typing $32=1 then click on the Send button will turn laser mode on, $32=0 will turn off.  
  
For the kinetics problem, I think it is not to Grbl to make the axis corrections calculations. Because of many differents geometric machines configurations, Grbl must stay as standard as possible to work on a maximum of differents machines. I think the kinematics calculations mus be in the post-processor wich will be unique for each machine.  
  
@++;  
Gauthier.

closed 2018-07-13 by fra589