fra589/cn5X #7: About language translation

cneve, 2018-09-26:
Hello, recently, I have encountered problems in language translation.
I copied a cn5X.en.ts file, and renamed it, and translated the characters using Linguist, and finally published the PM file using Linguist,which was saved in the same directory as the TS file.
The corresponding language selection is already shown on the software menu, but when I click the button, the command window appears "Locale () not usable, using default to english"
If I want to make a local language,What should I do next?

fra589, 2018-09-26:
Hi @cneve 
You also need to update the cn5X_locales.xml file in the i18n directory...
In wich language do you have made your translation ? It will be fine if you can send it to me :-)
@++;
Gauthier.

cneve, 2018-09-27:
I am now changing in the en.ts file. After I have finished, I will send you the TS file. Gauthier thank you for your help.

luisluna2, 2018-09-27:
@fra589 
Friend, how can traslate?, with what program?

cneve, 2018-09-27:
For example：
1、self.btnReset.setText("Soft Reset (^X)")         in  mainWindow.py
2、Copy this statement to “ def retranslateUi(self, mainWindow):” Method
3、Like this “self.btnReset.setText(_translate("mainWindow","Soft Reset (^X)"))” and save the mainWindow.py (Best use of Python IDLE editing)
4、Open cn5X.en.ts file
5、Add a statement
      <message>
        <location filename="../mainWindow.py" line="1852"/>
        <source>Soft Reset (^X)</source>
        <translation>Your translation</translation>
    </message>
6、Save the file cn5X.en.ts
7、Install QT linguist software，Open the file cn5X.en.ts，Release documents to cn5X.en.qm
8、Open the cn5X and complete the translation. lol

fra589, 2018-09-27:
> how can traslate?, with what program?

Hi @luisluna2,
You don't need to make any change in the Python .py files.
- First, install the QT linguist software,  
- Then, go to the \<cn5X\>/i18n directory,  
- Duplicate the file cn5X.en.ts to cn5X.es.ts for example to make the spanish translation (the actual cn5X.es.ts is an empty translated file, so you can replace it),  
- Open your new .ts file with QT linguist software,  
- Change the language cible in the "Edit/File parameter" menu,  
- Replace all the items english translation by your new langage translation,  
- Save the new translated .ts file,
- Use the "File/Publish" menu to publish the .qm file in the same \<cn5X\>/i18n directory. It is this cn5X.\<LOCALE\>.qm wich is used by cn5X to translate the interface at run time,  
- Don't forget to update the i18/cn5X_locales.xml to reflect your new translation,  
- Dont forget to send to me your new .ts file :-)  

Thanks for your contribution,

@++;
Gauthier.

fra589, 2018-09-27:
Hi all,

To avoid duplicate work, please, write here the translation you plan to do.

Thanks,
Gauthier.

cneve, 2018-09-27:
Gauthier ，if  translate some of the original code of grbl, will there be problems?Grblcom connection failure occurs several times when clicking connect.

fra589, 2018-09-27:
HI @cneve,  
If you translate the grbl code itself, cn5x to grbl serial com will stop to work since cn5X analyse and wait for the original sentences of grbl.  
For example, to complete the connection, cn5X wait to receive a string (case sensitive) wich begin with "Grbl " and end with "help]"... So, if you translate the word "help" in the original code of grbl, cn5X never end its connection.
@++;
Gauthier.

luisluna2, 2018-09-27:
Hi @fra589 I will modify to Spanish :)

dbxzjq, 2019-02-03:
hi @fra589 
 CN5X can publish WINX86 installation file, thank you!

fra589, 2019-02-05:
Hi @dbxzjq, 
I am sorry, but my Windows develop box is broken (pleonasm :yum:), so, I can't make Windows installation files for the moment(and probably for a long time).
But you can install prerequisites separately and run cn5X directly after...  
  
To install on Windows :
- Download and install latest Python 3 from https://www.python.org (don't forget to add Python in PATH environment variable)
- Run cmd (as Administrator if you have installed Python for all users)
- in cmd, type "pip install pyqt5"
- Download and copy the whole cn5X directory from Github,  
  
=> Double click on the cn5X.py downloaded file in Windows explorer... 

Enjoy using cn5X++ 

@++;
Gauthier.

closed 2021-02-22 by fra589