fra589/cn5X #8: Trouble running cn5X under Python 3.5.4

rodrigorrm, 2018-09-27:
Hello there,

It is probably more a heads up and nothing to be done here, but, PyQt5 has problems running under Python 3.5.2 and this prevents cn5X from even firing up.
The fix would be running Python 3.6.5  instead; the issue is discussed here: [https://github.com/mu-editor/mu/issues/575](url).

cneve, 2018-09-28:
Do you run cn5X on windows or Debian？when you test。
Cn5X-0.3-alpha can be opened on Windows32 bit system with python3 idle.

rodrigorrm, 2018-09-28:
Hi, @cneve, 
I'm trying to run it on Ubuntu 16.06 64bit. Maybe running on Windows would sort things out. 
I'll give Ubuntu some more attempts and will let you know of my findings.

cneve, 2018-09-28:
Looking forward to your new discovery.(*￣︶￣)

fra589, 2018-09-28:
Hi,
The development of cn5X++ is made under Debian testing and sid, with Python 3.6.6+ or 3.7.15+... Depending of wich box I used. The code is first tested on Linux, then on Windows 10.
@rodrigorrm,
Can you post the error message you get on Ubuntu ?
@++;
Gauthier.

cneve, 2018-09-29:
@fra589 
1、Can cn5X-0.3-alpha run in Debian system?
2、Can cn5X-0.1-alpha menu be translated into my language run in Debian system?
3、Cn5X-0.1-alpha runs in Debian system. All button images can not be displayed.

cneve, 2018-10-10:
@fra589

1、Message window display:

2018-10-10 09:57:08 : Info    : cn5X ++: Grbl initialized.
2018-10-10 09:57:10 : Error   : grblComSerial: Error sending data: timeout, err # = 12

2、And here is my $$configuration 

$0=10
$1=255
$2=0
$3=0
$4=0
$5=0
$6=0
$10=255
$11=0.020
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=0
$24=500.000
$25=635.000
$26=250
$27=1.000
$30=27
$31=0
$32=1
$100=314.961
$101=314.961
$102=314.961
$103=314.961
$104=314.961
$110=635.000
$111=635.000
$112=635.000
$113=635.000
$114=635.000
$120=50.000
$121=50.000
$122=50.000
$123=50.000
$124=50.000
$130=200.000
$131=200.000
$132=200.000
$133=360.000
$134=180.000

I really don't know how to solve this problem.Can you help me?

fra589, 2018-10-11:
Hi @cneve,  

I just push a new cn5X version with extended debug on serial com.  
Can you, please, update your version with this new one and do this :  
- Open cn5X,  
- Click on "Debug" tab,   
- Click on "Debug" button,  
- click on "Pause pooling" button,  
- click on "Connect" button,  

Post here the debug ouptput.

@++;  
Gauthier.

cneve, 2018-10-11:
2018-10-11 16:39:43 : Info    : grblCom: Starting grblComSerial thread.
2018-10-11 16:39:43 : Info    : grblComSerial running "grblComSerial" from thread #0x3834.
2018-10-11 16:39:43 : Info    : grblComSerial: comPort COM5 open, RC = True
2018-10-11 16:39:44 : Info    : cn5X ++: Grbl initialized.
2018-10-11 16:39:49 : Error   : grblComSerial: Error sending data: timeout, err # = 12
2018-10-11 16:40:03 : Error   : grblComSerial: Error sending data: timeout, err # = 12
2018-10-11 16:40:06 : Error   : grblComSerial: Error sending data: timeout, err # = 12
2018-10-11 16:40:14 : Error   : grblComSerial: Error sending data: timeout, err # = 12

fra589, 2018-10-11:
> 1, I think debug refresh is too fast. Can you set it up a bit slower?
This why I tell you to click on the "Pause poolilg" button **before** connecting with Grbl...

I pushed new small revision with little correction in com initialisation algorythme, and more debug...

Please, repost new debug tab output.

cneve, 2018-10-12:
cn5X++ (v0.3.e) : Starting debug.
grblCom.startCom(self, COM5, 115200)
grblComSerial.__openComPort(self)
grblCom.on_sig_connect(self, True)
grblComSerial: comPort COM5 open, RC = True
grblComSerial.__openComPort() : Buffer recu : "b"\r\nGrbl 1.1h ['$' for help]\r\n[MSG:'$""
grblComSerial.__openComPort() : Buffer recu : "b"H'|'$X' to unlock]\r\n""
grblCom.on_sig_init(self, Grbl 1.1h ['$' for help])
>>> $I\n
grblComSerial.__sendData(), T = 1539303366943.8772 : self.__comPort.waitForBytesWritten(12)
grblComSerial : Donnees envoyees, T = 1539303366944.8774
<<< [VER:1.1h.20180712:]
<<< [AXS:5:XYZAB]
<<< [OPT:VNM,35,255]
<<< ok
>>> \n
grblComSerial.__sendData(), T = 1539303366979.8792 : self.__comPort.waitForBytesWritten(12)
grblComSerial : Donnees envoyees, T = 1539303366980.8792
<<< ok

【The above is the result of debug display.】

【The following results are displayed in the command line window, and the main window is forced to close.】

c:\cn5X-master>python cn5X.py
Traceback (most recent call last):
  File "c:\cn5X-master\grblComSerial.py", line 336, in run
    self.__mainLoop()
  File "c:\cn5X-master\grblComSerial.py", line 261, in __mainLoop
    self.__sendData(toSend)
  File "c:\cn5X-master\grblComSerial.py", line 147, in __sendData
    self.sig_debug.emit(self.tr("grblComSerial : Erreur envoi des donnees : time
out, err# = {0}, T = {}").format(self.__comPort.error(), time.time() * 1000))
ValueError: cannot switch from manual field specification to automatic field num
bering
QObject::~QObject: Timers cannot be stopped from another thread

c:\cn5X-master>

fra589, 2018-10-12:
> 【The following results are displayed in the command line window, and the main window is forced to close.】 
> (...)
> ValueError: cannot switch from manual field specification to automatic field num
bering  

OUPS ! Just a typo error when adding the last debug message... Corrected... Sorry :-$  

The original bug of this thread (writing timeout) seems strange, since cn5X have writed all the data on the serial port, then the timeout come when all the work seems finish...  

I will make a new virtual Windows machine with your Python version (3.5.4) to try to reproduce it.  
Don't worry, this may take a little bit of time...  

@++;
Gauthier.

cneve, 2018-10-12:
@fra589 
Well done

Looking forward to the latest version, I pay attention to cn5X every day.

cneve, 2018-10-12:
【Now    It looks much better.】
cn5X++ (v0.3.f) : Starting debug.
grblCom.startCom(self, COM5, 115200)
grblComSerial.__openComPort(self)
grblCom.on_sig_connect(self, True)
grblComSerial: comPort COM5 open, RC = True
grblComSerial.__openComPort() : Buffer recu : "b'\r\nGrbl 1.1'"
grblComSerial.__openComPort() : Buffer recu : "b"h ['$' for help]\r\n""
grblCom.on_sig_init(self, Grbl 1.1h ['$' for help])
>>> $I\n
grblComSerial.__sendData(), T = 1539329286116.9001 : self.__comPort.waitForBytesWritten(12)
grblComSerial : Donnees envoyees, T = 1539329286121.9006
<<< [VER:1.1h.20180712:]
<<< [AXS:5:XYZAB]
<<< [OPT:VNM,35,255]
<<< ok
>>> \n
grblComSerial.__sendData(), T = 1539329286151.902 : self.__comPort.waitForBytesWritten(12)
grblComSerial : Donnees envoyees, T = 1539329286151.902
<<< ok

【The message window is also normal.】
2018-10-12 15:33:45 : Info    : grblCom: Starting grblComSerial thread.
2018-10-12 15:33:45 : Info    : grblComSerial running "grblComSerial" from thread #0x4060.
2018-10-12 15:33:45 : Info    : grblComSerial: comPort COM5 open, RC = True
2018-10-12 15:33:46 : Info    : cn5X ++: Grbl initialized.
2018-10-12 15:33:48 : Info    : Unlocking urgent stop.
2018-10-12 15:34:09 : Info    : Reading the file: C:/cn5X/grbl.gcode
2018-10-12 15:34:09 : Info    : 9 lines in the file
2018-10-12 15:34:11 : Info    : Starting cycle...
2018-10-12 15:34:14 : Info    : Not decoded Grbl response: [,27.620,8.810,-8.810,-17.620|Bf:26,255|FS:923,0|Ov:100,100,100>]

Now I can safely control stepper motors.      O(∩_∩)O

fra589, 2018-10-12:
Yes !  :ambulance: :checkered_flag: :ok_hand:  
  
It seems that the little correction in com initialisation algorythme that I made 2 reply ago (sequence order change) is the right solution.  
  
Now, I can close this issue.  
  
Don't forget to disable debuging for normal operation of cn5X++.  
 
@++;  
Gauthier.

closed 2018-10-12 by fra589