fra589/cn5X #9: Arduino Pinout

rieto, 2018-11-06:
Hello, thanks for sharing this project.

Is there an Arduino Pinout available? Like Probe Pin or Temp Sensor for Spindle Temperature for Fans on the water cooling radiator. Are you planning to make some features? Like: Glass Scale with an PID Controller with setpoint from Arduino actual value from Glass Scale  and Rotary encoder. If you are planning to do something like this please contact me im an electronic guy. And how can I install the GCode Sender Software on Windows?
Best regards from Germany/ Darmstadt
Tobi

hzxiaoyu, 2018-11-10:
PROBE pin is A15.

fra589, 2018-11-10:
Hi @rieto,  

The pinout definitions for the [grbl-Mega-5X project](https://github.com/fra589/grbl-Mega-5X) on RAMPS 1.4 is here :  
- https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout

Like @hzxiaoyu said, the probe functionality is implemented on A15 by default.  
There is no plan to add Temp sensor or other input for the moment.  
  
What do you mean by "Glass Scale ?"  

To install cn5X++ control panel / gCode sender on Windows, you need install [Python3](https://www.python.org/downloads/windows/) and [PyQT5](http://pyqt.sourceforge.net/Docs/PyQt5/installation.html). The PyQT5-QtSerialPort module is installed by default with PyQT5 on Windows.

When you have installed Python and PyQT5, add Python to the Windows system path and run the cn5X.py script.

Best regards from France / [Audenge](https://www.google.com/maps/place/33980+Audenge/@44.7132478,-1.0932864,11z/data=!3m1!4b1!4m5!3m4!1s0xd54962fc98c448d:0xfb5c14427666a488!8m2!3d44.683992!4d-1.015512)
@++;
Gauthier.

rieto, 2018-11-13:
Hi,
sry for my technical English. I mean this "https://www.youtube.com/watch?v=vHXqTl8NUww".
The actual Machine Position can directly send to the Computer/Pi and the Position where the machine should is calculatet by the Arduino. With another Arduino Nano or something like this per Axis we can Calculate the the difference and do the movements with an PID Controller. That way is much more acurate ( if you want).

hzxiaoyu, 2018-11-13:
@fra589 when you will update the cn5x with keyboard control axis? And the grbl communication is fresh so quick,I don't want see other  things except the command. What should I to do?

fra589, 2018-11-14:
Hi,

@rieto In Grbl, movements are generated by stepper motors and not by DC motors. It is therefore impossible to use a PID controller to drive the movement.
The calculated position to go is converted to number of steps, sent to the engines, and the result is returned in the cn5X interface. If there is no mechanical problem causing a loss of step, the position displayed in cn5X is the correct one.

@hzxiaoyu Keyboard control is planed and should be easy to implement (add a global keyboard hook and call existing jogging function when the right key is trigged).  
Unfortunatly, there are many other functionality that are needed for cn5X be fully functional (machine homing, G54 -> G59 & G92 offset control, etc.) and I have another job, other hobbies and only 24 hours by day and only one brain :-)  
So, I can't give you a delay for this. There are times when things go fast, sometimes, it's slower...

@++;
Gauthier.

closed 2018-11-14 by fra589