fra589/cn5X #11: Furst connection error

lukino87, 2018-11-18:
both firmware and software are at the latest version the error message occurs only during the first connection. starting first the serial monitor of arduino ide then its software instead the whole connects and works without any error.

cneve, 2018-11-20:
It is true that the first connection needs to be connected through a serial port monitor, and then cn5X can connect. So do I. @lukino87 @fra589

fra589, 2018-11-20:
Hello @lukino87, @cneve,  
This is strange !?!?!? Are you both on Windows ?  
In my home, it work fine from first connection, but usualy,  I use Debian Linux.  
Can you activate the debug tab of cn5X before connecting and post the whole debug output from the begining to the timeout message please ?  
@++;  
Gauthier.

lukino87, 2018-11-20:
hello and thanks again for your work and your help yes I'm on windows 10 64 as soon as I get home I will make you have the debug thanks again

fra589, 2019-02-03:
..No more info for debug... Issue closed.

closed 2019-02-03 by fra589