fra589/cn5X #12: Linear vs Rotary

rasciodc, 2019-01-16:
Hey Gauthier, how are you? Happy new year!

This is really not a issue but rather a quick question. As you must know, I've been using Grbl Mega 5X to operate a 4 axis hotwire cutter, which has 4 linear axis.

Can I use cn5X++ to operate all axis as linear or are axis A, B and C configured to be rotary only (as the picture/logo on main page suggests)?

The printscreen of the program on the Wiki shows the jogging buttons for A and B axis also suggesting they are rotary... are they hardcoded to move in degrees/radians only or would they work for linear motion too? 

Thanks!!!

fra589, 2019-01-16:
Hi Daniel, Happy new year to you too !

All axis can be linear, in config.h, use :  
```
#define N_AXIS 4            // Number of axes
#define N_AXIS_LINEAR 4 // Number of linears axis
```
And configure the motors step per unit using units as millimeters.
In cn5X, the buttons just send jog axis move with value, without interest to known if they are linear or rotating moves.  
  
@++;
Gauthier.

rasciodc, 2019-01-16:
This is great! Thanks!

Just a question now... when I set N_AXIS_LINEAR 4, which axis are contemplated? I assume it would logically follow the order X Y Z A B C.... is that correct?

fra589, 2019-01-17:
Hi Daniel,
When you set N_AXIS_LINEAR 4, the first 4 axis are considered linears. Ordered by axis number and not by axis_name : order = AXIS_1, AXIS_2, ..., AXIS_N regardless of the value of their AXIS_*_NAME.
@++;
Gauthier.

closed 2019-01-23 by fra589