fra589/cn5X #13: Install on RaspberryPi

entelzonta, 2019-02-13:
I command in ssh,

apt-get install python3 python3-pyqt5 python3-pyqt5.qtserialport
adduser pi dialout
git clone https://github.com/fra589/cn5X.git
cd cn5x/
make

and then getting some errors like this

`pi@raspberrypi:~/cn5X $ make
make[1]: Entering directory '/home/pi/cn5X/i18n'
lrelease cn5X.pro
make[1]: lrelease: command not found
Makefile:35: recipe for target 'cn5X.en.qm' failed
make[1]: *** [cn5X.en.qm] Error 127
make[1]: Leaving directory '/home/pi/cn5X/i18n'
Makefile:40: recipe for target 'lang' failed
make: *** [lang] Error 2
`
or how can i install and run this app

have a nice day

fra589, 2019-02-15:
Hi @entelzonta,  
You don't have to use make. The makefile is just for internationalization develop. Just run cn5X.py and it should work with the install you have alleready done.  
@++;  
Gauthier.

entelzonta, 2019-02-16:
Hi again

show in terminal like this

`pi@raspberrypi:~/cn5X $ python cn5X.py
  
File "cn5X.py", line 504
   
 def on_sig_config_changed(self, data: str):
                                                           ^
SyntaxError: invalid syntax
`

fra589, 2019-02-17:
Hi @entelzonta 

By default, on your Raspbery, python version is in version 2. So, to run a python 3.X script, you need to use :  
`$ python3 cn5X.py`

or : `chmod +x cn5X.py` and `./cn5X.py` wich use by default python3 to run itself (cf first line of the script).  

@++;
Gauthier.

entelzonta, 2019-02-17:
Hi,

Oh got it. Thank you solved my problem

closed 2019-02-17 by fra589