fra589/cn5X #14: All button disabled except for the Emergency Stop button

n0th1n6, 2019-03-15:
I have built APP_VERSION_STRING = "0.3.f" both on Windows 10 and Windows 7. Build was succesfull but the buttons are all grayed. I can only press Emergency Button. I can connect to grbl-Mega on a ramps 1.4 but the connection is problematic. I can confirm that it is not an issue with the arduino board or CH340 because I can complete a job using CNCJS. I will post screenshot once I'm back home.

[image]

Above is a screenshot in my win10 build but it is not connected to grbl-Mega. I change the version as it wont compile with a alpha character in the version.

fra589, 2019-03-15:
hi @n0th1n6,
This is a normal behavior. For security reasons, the CNC machine come in Emergency Stop status when power up and/or connecting. So unwanted movements can't come.
You have to "unlock" emergency stop by double clicking on it.
After unlocking, you will can use all functionalities, except configuration tool (Machine/Grbl configuration menu) wich require the Emergency stop status before use it, for the same security reasons.
@++;
Gauthier.

n0th1n6, 2019-03-15:
Hi @Gathier

I might have missed it when it enabled the buttons but believe me I have
done that already. Ill try later if I can make it work.

BR,

Noli

n0th1n6, 2019-03-16:
It is working now. Thanks @fra589

closed 2019-03-16 by n0th1n6