fra589/cn5X #15: Windows Installer - Updated Release

MeJasonT, 2019-04-10:
Hi Gauthier,
Somehow i managed to find a windows installer for cn5X and installed it on windows 7 64 bit OS.
At first it wouldn't connect with my Arduino Mega/controller, I read through the issues and read that there was a bug - CrLf had not been included when the code first connected with the mega and a fix was issued v0.3.i. I can only find the source code for v0.3.h (python based GUI ?).  Is it possible to obtain a windows installer with the fix.
I have a workaround i'm currently using which is to hold down the boards reset switch whilst pressing connect on the GUI this is putting the reset and connection in sync when the reset is released and is successful every time.  I currently use Grbl Panel which is unfortunately no longer supported and only 3 axis GUI and Candle cnc which is not as feature rich as Grbl panel also only 3 axis GUI.  I like your application as it has a look of realism about is, its nice to use a GUI which looks and feels like using an industrial machine operating panel.  Looking forward to your reply.
Jason

fra589, 2019-04-11:
Hi Jason,
Effectively, I corrected this connection issue on Windows last week, and I don't have published the Windows installer for this new version... Probably I will do it later, but I'm so busy, that I can't tell when...
But installer is not the optimum solution for use cn5X as it make a Windows executable from Python script wich embled all the Python and needed modules in one exe install.
I think it's better to install last updated version of Python from [python.org](https://www.python.org/downloads/) (don't forget to add python in your PATH, and .py file association that are install options)
Then use the pip command to install dependency : in Windows command prompt, type :
`pip install pyqt5`
After that, you can download the last version of cn5X from this github, copy or extract the whole dir where you want and run it by just double clic on the file cn5X.py.
I hope this help you,
@++;
Gauthier.

MeJasonT, 2019-04-11:
Hi Gauthier,
thanks for your reply, i have managed to find the pip install in an earlier post and have managed to open 0.3.0h form the releases.  I now realise i can download from the master branch which is your current up to date version.  I will try it tonight.  I have loaded python on my Machines before and not had much success - always complaining how many programming languages do i have to learn and had little success.  The pip install allowing windows users to point and click to open python files is pure genius.
Its nice when someone explains how to do things.
Many thanks for your support, its much appreciated.  Happy for you to close issue, the explanation is most helpful and i'm sure others will find the same.

closed 2019-04-12 by fra589