fra589/cn5X #17: Hard limits and A axis display bugs?

7b-w, 2019-12-29:
I am using cn5X++ V 0.3k and have enabled hard limits via $21=1 and when I press a limit switch I can see the round limit light at the bottom of the display window light up so i know 5X++ is reading the limit switches yet no alarm shows and the axis keeps moving. Is there something else that needs to be setup for this to work? So i am using the A axis and it is working but the display is not showing degree's and looks more like a linear measurement, again is there a setting that needs to be setup for this to display right? So last item is when jogging when I get to zero on the display position on X or Y (not A that works OK) the movement stops and won't go any farther but i can go in the opposite direction OK, settings?
Thanks oh and if there is a better place to ask these questions please let me know.

7b-w, 2019-12-30:
So I found if I changed the display units from inch back to mm then the A axis displays in degrees. $13=1 to $13=0 took care of the A axis display issue.

7b-w, 2019-12-30:
Looking further into the hard limits issue I found that hard limits for Ramps PCB is not enabled so will have to change that to fix the limits issue.

fra589, 2019-12-30:
Hi @7b-w,  
  
Hardware limits are working only in the last (1.1m) version of grbl-Mega-5X. You need to enable the `#define RAMPS_HW_LIMIT` in the grbl-Mega-5X's cpu_maps.h file (line 321) to be able to use it.  
  
For your last item, this is probaly due to the soft limit ($20) config of grbl (see https://github.com/fra589/grbl-Mega-5X/wiki/Grbl-Configuration).
After homing, Grbl will set by default the entire machine space into negative space, as is typical for professional CNC machines. So, if you use the soft limit, movements will be restricted between 0 and -AXIS_MAX_TRAVEL. soft limit are ignored if AXIS_MAX_TRAVEL is 0.  
You have to define your machine and / or piece coordinate systems to use positives and negatives coordinates. see http://linuxcnc.org/docs/2.7/html/gcode/coordinates.html for more.

For the A axis output, it's a bug in grbl-Mega-5X outputs when outputs are defined in inches, and not a bug in cn5X++. I will try to correct it ASAP...  

@++;
Gauthier.

7b-w, 2019-12-30:
Gauthier,
Thanks for taking the time in replying to these issues about your software, 
I will have to look at what AXIS_MAX_TRAVEL is set at but soft-limit $20 is set to 0. I will check out the links you have provided.
Happy New Year!
Bruce

7b-w, 2019-12-30:
So I had time to look at the last link on coordinates.html and was wondering why on the control panel I see G54 to G59 but not G53? I have not set any offsets to where X0 Y0 might be set. Ok now this brings up a question, say I jog the X and Y axis to some point and now want to set the origin to that spot so X=0 and Y=0 I don't remember any button that "sets zero" on the control panel. What am I not seeing?
Thanks,
Bruce

fra589, 2019-12-30:
Hi Bruce,
cn5X++ is still in active development, documentation is not written :-)
To define G54, G55, ... G59 offset, active GXX you want by clicking on it, then jog (or move with G0/G1 command in the GCode text area) to the desired origin for one or all axis.
Then, right click on the axis you want to define origine and chose "reset <X,Y,Z,...> axis value to zero" or "reset all axis values to zero".
But I agree, it is not obvious and I need to make a better interface for this!
I think this will be done soon because I began to use my real CNC with my 5X* softwares and then, I see all ergonomics difficulties and bugs by myself :-)
The G53 is not on the control panel because there is not offset definition for it and G53 is not modal like G54 to G59. To use the absolute machine coordinate system G53 moves, you need to specify G53 on the same line as a linear move. (see http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g53)
Effectively, cn5X++ need a specific interface to use machine coordinate system moves.
@++;
Gauthier.

7b-w, 2019-12-31:
Gauthier,
I understand about G5X codes and thanks for definition and software still in active development but so far I like what I see.
I tried the right click on axis display and tried the selections. Reset axis to zero does not work but move axis to zero does. I did not try reset all axis to zero.
Another issue I found was just moving the X axis say 10 mm from origin or -10 mm from origin the X axis would move in only one direction minus, plus the display doesn't show the correct value (never 10 something like 9.889 ) yet the Y axis moved correctly and displayed correctly.
Back to jogging, I can jog in minus direction from plus value to zero and continue in minus direction but when I move from minus position to zero it stops at zero and will not continue on past into positive area. Soft limits and homing are set to 0.
My setup is for laser engraving and got the PWM working Ok with S gcode commands for laser output power.
Looking at cpu_map.h file I see no commented out line (line 318 in your info) but do see limit max and limit min commented out. Will have to give you more info on that later. 
Happy New Year to you and 73's
Bruce

fra589, 2020-01-01:
Hi Bruce, Happy new year !  
  
Warning, there is 2 groups of items in the axis position context menus:  
One group to define G53-G59 coordinates offset (or piece origin) labeled "Reset ? axis to zero"
and the other group to move (jog) to the origin, labeled "Return of ? to the zero position"
[image]
Be aware to use the first group of items to define the G53--G59 origin offset.
I think I have to use more explicit label for those context menus.
  
If your X axis move always in the same direction and don't change when you ask him to positive or negative direction, it's probably due to an hardware problem on the X direction pin...
9.889mm in place of 10mm can be possible depending of the step/unit configuration. Grbl can only move to a integer number of motor steps. So, it is be possible that it can't go to your exact position.
Can you post the **`$$`** output? So I will can explain better your behaviors :smiley: 

If you don't have the `#define RAMPS_HW_LIMIT` at line 321 of cpu_map.h (and the comment about this feature at 318), it's probably due that you don't have the last version of grbl-Mega-5X: v1.1m.20191215.
Can you post the **`$i`** output of grbl to verify?
  
For your last jogging issue, this is due to a configuration that is not implemented in cn5X++ when you don't use software limit and origine definition in Grbl.

When you power on Grbl, he can't know where are the axis, and he make all axis values to 0. If you don't use homing, you should also disable software limits to use positive values without defining origin. Grbl neither cn5X can know where are the limits !
Unfortunately in your case, cn5X needs to know the possible limits of movements in absolute coordinates of the machine to be able to make a continuous jog up to these limits. If he can't calculate them ($ 20 = 0 and $ 22 = 0 or homing canceled), he use the norm of 0 to -MAX_TRAVEL. So, cn5X can not use continus joggin in the positive absolute coordinate space. Unless to use not null jog increment, I don't have a solution for this problem today. I will have to think about this case to find the least bad solution... The best is still to define the limits of the machine and use the homing features.  
  
@++;  
Gauthier.

7b-w, 2020-01-01:
Here is $$ and $i outputs:

Grbl 1.1l ['$' for help]
$0=10
$1=255
$2=0
$3=3
$4=0
$5=0
$6=0
$10=255
$11=0.020
$12=0.002
$13=0
$20=0
$21=1
$22=0
$23=0
$24=500.000
$25=2000.000
$26=250
$27=5.000
$30=1000
$31=0
$32=1
$100=80.000
$101=80.000
$102=80.000
$103=8.889
$104=8.889
$110=9000.000
$111=9000.000
$112=300.000
$113=1440.000
$114=1440.000
$120=300.000
$121=300.000
$122=100.000
$123=100.000
$124=100.000
$130=200.000
$131=200.000
$132=200.000
$133=360.000
$134=180.000
ok
[VER:1.1l.20190605:]
[AXS:5:XYZAB]
[OPT:VNM,35,255,10]
ok
ok

7b-w, 2020-01-01:
Where can I find v1.1m.20191215?

X axis
I think I don't understand G54 and the offset it uses so need to read up on that.

I am so use to using mach3 and Flashcut and what those let me do when it comes to jogging this new control panel has me a bit confused.

I really need to update grbl 5x in the mega 2560 to the latest.
Bruce

fra589, 2020-01-01:
This is what I sayed, you don't have the latest 1.1m.20191215 grbl-Mega-5X version, so, hardware limits are not implemented in your version.
I have not published yet a compiled binary version of it.
You can download the latest source version on his Github page : https://github.com/fra589/grbl-Mega-5X and compile it by yourself with Arduino software cf [original grbl wiki procedure to compile Grbl](https://github.com/grbl/grbl/wiki/Compiling-Grbl)
You will need to compile it to ajust the functionalities you need like hardware limits enabled.
If you can't, ask me and I will do it for you and I will post the hex file...

This is the detail of grbl incremental movement calculation that explain it can be aproximative:
With 80 steps per mm ($100), your X axis is able to go exactly to 10mm (it need 800 steps exactly to do this). 
On the other hand, axis A or axis B, with 8.889 steps per degree ($ 103 & $ 104) will go to the nearest position for 10 units, that is:
Closest steps = round_integer (10 / (1 / 8.889)) = 89
Route taken by these 89 steps = (1 ÷ 8.889) × 89 = about 10.012

For G54 and offset knowledge the best is to read the link above on Linux CNC.

@++;
Gauthier.

7b-w, 2020-01-01:
So doing a search I found what I thought might be 1.1m20191215 and I was right so compiled that using Arduino IDE and downloaded ok. Oh I un-commented out the Ramps_HW_limit and limits work but trying to move off the limit that is active doesn't happen. I have to physically push the axis off the switch then disconnect and reconnect serial link. Hitting soft reset and unlock alarm isn't working unless again i am doing or setup something wrong.
Here is $$ output now:
Grbl 1.1m ['$' for help]
[VER:1.1m.20191215:]
[AXS:5:XYZAB]
[OPT:VNMH,35,255,10]
ok
$$
$0=35
$1=255
$2=0
$3=3
$4=0
$5=0
$6=0
$10=255
$11=0.020
$12=0.002
$13=0
$20=0
$21=1
$22=0
$23=0
$24=500.000
$25=2000.000
$26=250
$27=5.000
$30=1000
$31=0
$32=1
$100=80.000
$101=80.000
$102=80.000
$103=8.889
$104=8.889
$110=9000.000
$111=9000.000
$112=300.000
$113=1440.000
$114=1440.000
$120=300.000
$121=300.000
$122=100.000
$123=100.000
$124=100.000
$130=200.000
$131=200.000
$132=200.000
$133=360.000
$134=180.000

Thankyou for all your help so far, I should write a simple gcode program and try it out to see how things behave.
Bruce

fra589, 2020-01-03:
@7b-w said:
> Oh I un-commented out the Ramps_HW_limit and limits work but trying to move off the limit that is active doesn't happen. I have to physically push the axis off the switch then disconnect and [...]

As explained in the [original grbl wiki about hardware limits](https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#21---hard-limits-bool):

> Keep in mind, that a hard limit event is considered to be critical event, where steppers immediately stop and will have likely have lost steps. Grbl doesn't have any feedback on position, so it can't guarantee it has any idea where it is. So, if a hard limit is triggered, Grbl will go into an infinite loop ALARM mode, giving you a chance to check your machine and forcing you to reset Grbl. Remember it's a purely a safety feature.

So, you effectively need to manually disengage the limit switch then reset and re-homing grbl to restart.
This is the reason I prefer software limits to secure Gcode out of limits errors, hardware limits are good for physicals problems and nothing else.

@++;
Gauthier.

closed 2020-02-06 by fra589