fra589/cn5X #18: Help to run the program

MARIOBASZ, 2020-02-06:
Sorry for my ignorance, how do I test your software?
1- I will download PYQT5, I will try to place it where it belongs (a little silly man from this side)
2-Which command should I execute? In others I find it easy: batch file .....
I hope help, hugs, Mario

fra589, 2020-02-06:
Hi @MARIOBASZ,
Do you try cn5X++ on Windows or on Linux ?  
If you try on Windows, tell me what is uncomplete on the wiki https://github.com/fra589/cn5X/wiki/Install-cn5X-on-Windows to get it running...  
@++;
Gauthier.

Gerhardathub, 2020-02-29:
Good day.
Im having trouble opening cn5x on my laptop windows 8.1. Not familiar with python.following your wiki steps doesnt work for me.  Can you please take me step by step through the process or even a video link. Thanks in advance

fra589, 2020-03-01:
Hi @Gerhardathub,  
<< _doesnt work for me_ >> Is not à bug, and does'nt tell me what's wrong on your config...  
- The first step of Wiki tell you to install last updated version of Python from [python.org](https://www.python.org/), and during the install process, add python in your PATH, and .py file association that are install options.
=> Did the python installation have worked fine for you ?
- The second step teel you to use [pip](https://pypi.org/project/pip/) to install dependency by typing `pip install pyqt5` in the windows command prompt.
=> Did you do it ? Did pip install pyqt5 dependency fine or did you have any error during the process installation ?
- The third step is to download and copy the cn5X whole directory on your computer and double-click on the cn5X.py file.
=> If the precedents steps have been done without errors, it should be work.

If no, try to open a Windows command prompt, then type:  
```
> cd <directory_where_you_downloaded_cn5X>
> python cn5X.py
```
You should have an error message or info wich will be usefull for debuging...

@++;
Gauthier.

Gerhardathub, 2020-03-01:
Good day

Python installed no error.
When i run the pip command get error windows cannot find pip.

Thank you

fra589, 2020-03-01:
Hi @Gerhardathub,  
  
You probably miss to check the "Add Python 3.X to PATH" wich is unchecked by default on the first panel of the Python install process...
[image]

Or perhaps have you unchecked the "pip" on the next install panel "Optional features" ?
[image]
  
@++;  
Gauthier.

Gerhardathub, 2020-03-04:
Hallo.

Python installed successfully
Pip install pytq5 installed successfully
Double click on cn5x unsuccessful. Opens black screen popup then closes
immediatly. Tried to run command prompt. No luck.

fra589, 2020-03-04:
Hi,
When you open a Windows command prompt then type :
```
C:\Users\Gauthier>cd <the directory where you downloaded cn5X>
C:\Users\Gauthier\src\cn5X>python cn5X.py

```
Do you have any error message ?

Gerhardathub, 2020-03-04:
Hello

After c:\users\gerhardus\downloads\cn5x\-0.1-alpha\cn5x-0.1-alpha\cn5x.py".
Line
26. In <module>
From Pyqt5 import QtCore. Qtgui. QtWidgets
ModuleNotFoundError :No Module Named `PyQt5`

Thats what i get.

fra589, 2020-03-04:
Hi  
We are going forward :-)
Your error mean that "pip install pyqt54" did not work as expected...
I think you have made an install of python for all users on your system (this is an install option of Python) and if so, the "pip install" command need to run in an administrator cmd for installilg PyQT5 also for all users.
So, open cmd.exe as adminidtrator then ret-run the "`pip install pyqt5`" command should correct the problem.  
@++;  
Gauthier.

Gerhardathub, 2020-03-04:
Hello.

 I found a way to run the program. I opened
PYTHON IDLE>file>open>Downloads>Cn5x-0.3.h>cn5x.py and hallo amazing app.
Thank you your help is much appreciated.

Regards

Gerhard Oosthuizen

Gerhardathub, 2020-03-04:
Hello.

New problem. Cn5x opens.
Port com4 usb serial ch340 connected.
Connected then gets error message
Error grblComserial Grbl initialization timeout
Error grblcomserial unable to open serial port
Grblcomserial End.
Cant control machine only button available is emergency stop

fra589, 2020-03-05:
Hi @Gerhardathub,  

It's like you haven't the last version !  
On your precedents posts, I can see you have the 0.3.h and the up to date version on Github is 0.3.k (3 release later) verify with the help/about dialog of cn5X...
[image]
The same problem about timeout under Windows have been discussed and corrected some time ago : https://github.com/fra589/cn5X/issues/11 linked to https://github.com/fra589/grbl-Mega-5X/issues/41
You need to download the last up to date version on the main cn5X++ page on Github: https://github.com/fra589/cn5X 
[image]
Use the green button to download the good version. The other version in the release page are obsolete. I need th search how to remove them :smirk:...  
  
About the only emergency stop wich is enabled at program start, this is normal for security reasons. See https://github.com/fra589/cn5X/issues/14
  
It's a always a goot idea to seach and read old issues about softwares :blush:...

@++;  
Gauthier.

Gerhardathub, 2020-03-05:
> linked to fra589/grbl-Mega-5X#41

Hello

Im not finding the v0.3k. Only these

Thank you

fra589, 2020-03-05:
??? Where do you download it ???
The 0.3.k is the one on the main cn5X Github page ! Not on releases page !!!

Gerhardathub, 2020-03-05:
Hi

Sorry a mistake from my side.
Managed  to download main page file.

Will run and test asap.

Do you know any good 4 to six axis grbl cam software for gcode toolpathing?

closed 2020-03-28 by fra589