fra589/cn5X #19: Testing cn5X on Windows 10

sns5400, 2020-07-10:
Hello,

Afther installation python 3.8 and PyQt5 in can start the interface but there is a communication problem.
I have setup a mega 2560 bord with grbl 5Mega firmware and still receive a time out error.

So nothing function.

I did the same test on a linux machine and the mega 2560 bord work normal.

Have you any tips ?

fra589, 2020-07-14:
Hi @sns5400,  
As cn5X is developed under Linux, I realized this problem, but not long ago.  
It seems that once again, a Windows update broke certain functions of the QSerial component of QT5 which I use for communication :-(  
As this is not the first time this has happened, I plan to rewrite the communication module without using the QT component. I think to get down to it after the summer... Maybe around September or October...  
In the meantime, I have no solution to offer for Windows...  
Sorry!  
@++;  
Gauthier.

sns5400, 2020-07-15:
ok thanks for your reply 

When it is windows maybe i find a work arround because on my linux pc it seems to work normal. 
I am bulding a 5 axis cnc machine and search for a controller so i came on your solution.

LETARTARE, 2020-07-15:
Bonjour,
j'avais eu ce problème avec 'grblQ-Mega-4axes', aussi j'avais récupéré l'ancien 'qextserialport' que j'ai modifié pour avoir le signalement des connexions et déconnexions avec 'Windows' et surtout 'Linux'. Je l'utilise depuis plus de deux ans : vous pouvez le trouver en [https://github.com/LETARTARE/qextserialport](url)
Si cela peut vous être utile.

sns5400, 2020-07-15:
Thank you for the tip 
I goig to try this tomorow 

Pierre

fra589, 2020-12-29:
Hi all,  
  
New version v0.5.a pushed today.  
This new version use [pySerial](https://pypi.org/project/pyserial/) in place of QSerial which dont work correctly for me in Windows.  
Warning ! you need to install the pySerial module before to use cn5X++.  
You can do this with pip cmd:
`# pip install pyserial`    

Since I'm in vacations and I don't have Windows box for testing, please, report if it work weel under it.  

@++;  
Gauthier.

closed 2021-02-08 by fra589