fra589/cn5X #20: Missing a G92 button..

spiderdab, 2020-08-26:
Hi, I found this grbl controller very useful, easy to read and stable. (I have problem both with UGS and bCNC, under PopOs!)
One thing that I'm missing is a Button to send a "G92 X0 Y0 Z0" like a "Set Zero", maybe with the possibility to fill 3 textfields if you want to set a position different from Zero..
Is this something you can add?
Thank you.

fra589, 2020-08-27:
Hi @spiderdab,  
  
This is a good question!
cn5X ++ is still in development, and the functionality you request is planned.
I unfortunately don't have enough time for the moment to do everything I hope to do, I think to be able to move this subject forward next winter...  
  
In the meantime, it is possible to use zero offsets G54 to G59 using the context menu with the right mouse button on the display of the working positions:  
[image]  
This send 'G10 L2 P- ...' gCode sentence to set the active G54, G55, ... or G59 coordinate system.  
  
@++;  
Gauthier.

spiderdab, 2020-08-27:
Thanks for answering. I didn't know about the right-click context menù.
You already did a great job.

fra589, 2021-02-18:
Hi all,  
  
G92 features have been added in addition to the G54-59 features in the today's new version 0.8.  
See in the Machine menu.

@++;
Gauthier.

closed 2021-02-18 by fra589