fra589/cn5X #21: CN5X locks up

Robin2g, 2020-09-19:
I have installed CN5X on Linux Mint and I have uploaded Mega-5X to my Mega.

The Mega seems to be working properly - I can send it commands with the Minicom terminal program.

When I try CN5X it allows me to use the X and Y jog buttons but if I press any other button on the panel (for example the Z jog) everything stops working but with no error message. I can no longer use the X or Y jog and I have to disconnect and reconnect to get the X and Y working again.

What might I be doing wrong?

...R

Robin2g, 2020-09-19:
On further investigation it looks like the problem may be in the Arduino code as CN5X seems to work fine with an Uno that has the regular GRBL uploaded.

When sending commands with minicom the grbl-mega-5X does not respond at all some times and then has to be reset. I presume it is this lack of response that is causing CN5X to get stuck

Very frustrating.

fra589, 2020-09-20:
Hi @Robin2g,  
  
I agree that the problem seems to come from grbl-Mega-5X...  
Which command is causing Grbl to freeze ?  
Can you post the result of `$I` and `$$` grbl commands  ?  
  
@++;  
Gauthier.

fra589, 2020-09-20:
Continue this conversation on [your other issue on grbl-Mega-5X](https://github.com/fra589/grbl-Mega-5X/issues/141)...

closed 2020-09-20 by fra589