fra589/cn5X #22: cnc5X on Debian

BosniaCNC, 2020-09-30:
Help to run the program on Linux (Debian-9).
Thank you very much!

fra589, 2020-10-01:
Hi @BosniaCNC,  
  
cn5X++ is developped under Debian, so, it's very easy to use it on another Debian
  
1) Install prerequisites:  
`apt-get install python3 python3-pyqt5 python3-pyqt5.qtserialport`  
2) Add the user to the unix group "dialout":  
`adduser <username> dialout`  
3) Download or clone cn5X and put it where you want on your system, ensure that the file cn5X.py is executable or make it executable:
`cd <cn5X_directory>; chmod +x cn5X.py`  
4) Run cn5X:
`cd <cn5X_directory>; ./cn5X.py`  

[image]

5) In the "**Port**" drop down list, chose the port where Grbl is connected, then click on the "**Connect**" button,  
6) Go to the "**Machine/Grbl configuration**" menu to adjust your Grbl parameters  
   or Double click on the "**Emergency stop**" button to unlock it and begin to use your CNC.  

@++;  
Gauthier.

BosniaCNC, 2020-10-01:
Thanks for the detailed instruction!

closed 2020-10-01 by fra589