fra589/cn5X #24: Enter on grbl configuration close cn5x (Raspberry PI)

fus3, 2020-12-13:
Hi, I recently (yesterday) installed cn5x on raspbian and was jogging with all axis and changing some things on configuration and every worked perfectly, but today i reinstalled raspbian and again installed cn5x but now everytime i connect the mega i have some messages of error:

2020-12-13 23:19:02 : Info    : grblCom: Starting grblComSerial thread on ttyUSB0.
2020-12-13 23:19:02 : Info    : grblComSerial running "grblComSerial" from thread #0x6a567460.
2020-12-13 23:19:03 : Info    : grblComSerial: comPort ttyUSB0 open, RC = True
2020-12-13 23:19:04 : Info    : cn5X ++: Grbl initialized.
2020-12-13 23:19:04 : Error   : grblComSerial: decode error: <class 'UnicodeDecodeError'>
2020-12-13 23:19:04 : Error   : grblComSerial: decode error: <class 'UnicodeDecodeError'>
2020-12-13 23:19:04 : Error   : grblComSerial: decode error: <class 'UnicodeDecodeError'>
2020-12-13 23:19:04 : Error   : grblComSerial: decode error: <class 'UnicodeDecodeError'>

 and then it connect. But if i exit from alarm state, cn5x only shows 3 axis (but it has 5 axis, because yesterday i could use the 5 axis and also if i send $$ to console then it shows grbl configuration settings for 5 axis) also if press mergency button to change the grbl configuration directly from cn5x i get out from cn5x and the terminal shows those messages

pi@raspberrypi:~ $ python3 /home/pi/cn5x/cn5X.py
qt5ct: using qt5ct plugin
cn5X++ running from: /home/pi/cn5x

                ####### #     #
  ####   #    # #        #   #     #       #
 #    #  ##   # #         # #      #       #
 #       # #  #  #####     #     #####   #####
 #       #  # #       #   # #      #       #
 #    #  #   ## #     #  #   #     #       #
  ####   #    #  ####
qt5ct: D-Bus global menu: no
qt5ct: custom style sheet is disabled
**Traceback (most recent call last):
  File "/home/pi/cn5x/grblConfig.py", line 325, in on_sig_config
    self.__di.lblGrblDate.setText(decodeVer[2])
IndexError: list index out of range
Aborted**

fra589, 2020-12-14:
Hi @fus3,  
  
Thanks for transmitting this little bug...  
It was due to your grbl doesn't have the standard grbl-Mega-5X version string structure (<major>.<minor>.<build date>).  
I just pushed a new version of cn5X++ which workaround this problem.  
  
@++;  
Gauthier.

fus3, 2020-12-14:
Hi Gauthier now i installed the new version you uploaded and the error persist but this time it shows another line of the grblConfig.py file. Output when entering on grbl configuration:

Traceback (most recent call last):
  File "/home/pi/cn5X/grblConfig.py", line 330, in on_sig_config
    self.__di.lneEEPROM.setText(data[1:-1].split(":")[2])
IndexError: list index out of range
Aborted

If i delete the line 330 of the file then the cn5x grbl configuration works and shows that my arduino mega 2560 has the 5 axes but there are some weird things on the initialitzation sub menu of grbl configuration.

fus3, 2020-12-14:
Also, about what you said _"your grbl doesn't have the standard grbl-Mega-5X version string structure"_ , I flashed grbl-mega-5x on my mega2560 about a month ago, so should i try to flash with the current version?

fra589, 2020-12-15:
Hi @fus3,  
  
OOPS! I think I publish this correction too rapidly and forgot another detail...  
I will correct it tomorrow...  
It's strange that your grbl-Mega-5X does'nt send it's standard version and information string. Can you post the result of the $I Grbl command?  
  
@++;  
Gauthier.

fus3, 2020-12-15:
Ok, perfect, about the result of the $I Grbl command:

On "cn5X++ messages" menu:

2020-12-15 13:36:17 : Error   : grblComSerial: decode error: <class 'UnicodeDecodeError'>
2020-12-15 13:36:17 : Error   : grblComSerial: decode error: <class 'UnicodeDecodeError'>
2020-12-15 13:36:17 : Error   : grblComSerial: decode error: <class 'UnicodeDecodeError'>
2020-12-15 13:36:17 : Error   : grblComSerial: decode error: <class 'UnicodeDecodeError'>
2020-12-15 13:36:17 : Info    : Not decoded Grbl response: [L,35,255,16]]

on "Grbl communication" menu:

$I
L,35,255,16]
ok

fra589, 2020-12-15:
Hi @fus3  

I just pushed a new version with error trapping which ignore this.

But your grbl-Mega-5X seems to have a communication problem!
The reply to the $I Grbl command should be of the form:
```
[VER:1.1p.20201203:]
[AXS:5:XYZAB]
[OPT:VNMGH,35,255,48]
ok
```
(this is the standard reply from the latest version)

You probably should flash the last version of grbl-Mega-5X to your board and reset all the EEPROM with the `$RST=*` command.
After this, you will have to replace your configurations parameters.  
  
@++;  
Gauthier.

fus3, 2020-12-15:
OK i tried your new version and it worked perfectly now i flashed the mega with the last version of grbl-mega-5x and restored defaults using $RST=* and now $I and shows: 
[VER:1.1p.20201203:]
[AXS:5:XYZAB]
[OPT:VNMGH,35,255,48]

Thank you for everything Gauthier!

closed 2020-12-16 by fra589