fra589/cn5X #27: Setup.py Syntax error

stoky996, 2021-02-11:
Hi 

when i try to run the setup.py script it errors out with a syntax error on this line and it is on the latest version from the github.

Syntax errors detected :

Line 62:
options = dict(build_exe = buildOptions),
^
SyntaxError: invalid syntax

PS i am a complete novice with coding atm

regards jesse

fra589, 2021-02-11:
Hi @stoky996,  
  
Thanks for reporting this issue, it was just a comma missing, Oops! I fixed it.
In fact, setup.py is useless for the operation of cn5X ++, nor is it used for installing. I had started to create it in order to provide a version compiled by cx_Freeze, but this part is not finished yet and the compilation is not operational.
- To install cn5X++ on Windows, there is a very simple procedure described here:  
https://github.com/fra589/cn5X/wiki/Install-cn5X-on-Windows  
- For installation on Linux, it's here:  
https://github.com/fra589/cn5X/wiki/Install-cn5X-on-Linux  
  
@ ++;  
Gauthier.

stoky996, 2021-02-11:
Thanks for fixing it yeah all tutorials on python that I saw during a quick Google search for install python scripts said to use setup.py thanks for the installation guides will try when I get home from school

stoky996, 2021-02-11:
Just tried running the cn5x.py script was not working looked at it in ide there is now only a blank file in it's place looked at the older one of have on another computer and it has code in it so can you please try re uploading the code

stoky996, 2021-02-11:
Hi just got it to work on my pi overwrites every old file with new except the Cn5X.py file all good now

closed 2021-02-17 by fra589