fra589/cn5X #30: Ramps 1.6

boxgame9, 2021-03-20:
Hi Gauthier
Can you confirm that mega 5x works with the ramps 1.6 board please

Pouvez-vous confirmer que mega 5x fonctionne avec la carte des rampes 1.6 s'il vous plaît

Regards Lesli

fra589, 2021-03-21:
Hi Lesli,  
  
Yes, it work with RAMPS 1.6.
J'utilise une carte RAMPS 1.6 sur l'une de mes plateformes de test.
  
@++;  
Gauthier.

boxgame9, 2021-03-21:
Thanks for the quick reply Gauthier.
Merci pour la réponse rapide Gauthier.
Stay safe and a good weekend
Restez en sécurité et passez un bon week-end

boxgame9, 2021-03-26:
Hi Gauthier,
Bonjour Gauthier
Although I think it unlikely will the mega 5x work on the ARM based mega 2560 please.
Bien que je pense qu'il est peu probable que le méga 5x fonctionne sur le méga 2560 basé sur ARM, s'il vous plaît.

Have a great weekend.
Passe un bon weekend.

Lesli

fra589, 2021-03-27:
Hi Lesli,  
  
I think you want to tell about 32 bits Arm based microcontroller with the same form factor than the Arduino mega 2560 which is the Arduino Due...
  
grbl-Mega-5x is not compatible with the Arduino Due.  
  
A while ago I was thinking of porting grbl-Mega-5X to the arduino Due, but I gave up on that idea for many reasons.  
There is many other 32 bits Grbl versions, and I like the idea of keeping the Mega 2560 card running. It's a simple card, really cheap and easy to get hold of. It's ideal for beginners and good enough for the hobbyist that I am.  
  
@++;  
Gauthier.

closed 2021-03-27 by fra589