fra589/cn5X #32: $10 Setting

HLex79, 2021-07-28:
Hi Gauthier!

First let me say a big thank you for your awesome soft! It runs smooth on Linux, however I was not able to have it performing well on Windows.

I have a question on the $10 setting, i have read the wiki an i can not figure out the $10=2 value with the buffer?

What should one use, i presume =1? I have strange black boxes appearing in the DRO field recently while hovering over with the mouse cursor.

BTW is there a setting =0 for this value?

Thanks a lot,

Lex.

fra589, 2021-07-29:
Hi Lex,  
  
The $10 parameter is used to define the elements returned by the "?" Command.  
  
This is a binary mask where only the first 2 bits are used.  
  
- Bit 0 defines whether you want to receive the working position or the machine position: If the bit value is 0, "?" returns the working position (WPOS), if it is at 1, "?" returns the machine position (MPOS).  
- If the bit 1 value is 1, the state of the Grbl's internal buffers (planner and serial RX available buffer) are added to the result of the "?" Command.  
  
The cn5X++ program itself manages and force the value of the parameter $10: Depending on the choice of the "Machine / View working positions (WPOS)" or "Machine / View machine positions (MPOS)" menu.  
Bit 1 is alwais set to 1 so that cn5X++ can know the state of the Grbl buffers.  
  
@++;  
Gauthier

HLex79, 2021-07-30:
Hi Gauthier,

Thanks for your explanations! I have set it back to zero and cn5X is back as new 😁
Best,

Lex

closed 2021-08-03 by fra589