fra589/cn5X #33: Does not work for me under Windows 10

jdeline, 2022-01-23:
Hi.  I see recent updates to files in the github repository, so I presume you are still working on it.  I'd like to share my experience.  I am running under Windows 10 and Python 3.9.  I was getting errors from python about a missing method to the serial module called "to_bytes", but I was able to overcome that.  I was then able to run cn5X and connect to a Mega 2560 running the latest version of Grbl-Mega-5X.  However, although connected, everything is greyed out and unusable.  By everything, I mean jogging, probing, and even entering gcode commands manually.  I am using a copy of grbl-mega-5x that I compiled with hard limits disabled and homing not required, so when connected to cn5x it starts up in an "idle" state--but everything is still greyed out.  If you need more info please let me know.  Thanks.

fra589, 2022-01-24:
Hi @jdeline,  
  
Yes, I still work on grbl-Mega-5X and cn5X++ softwares.  
  
The TODO list is still pretty long, and I apologize, but I don't find enough time to write the full documentation of cn5X and other things.  
  
When cn5X is launched and when it connect to Grbl, for security raison, the emergency stop button is locked in the off position.
To unlock it, and so, unlocking the CNC functionalities, you must do double-click on it.  
With a real Urgent Stop button, a single press lock it, you must turn it to unlock. With the software button in cn5X, a single click lock it, you must to double-click to unlock.  
  
There is another subtlety of the same style in cn5X++:  
The dialog box allowing you to modify the Grbl's configuration parameters is only accessible when the emergency stop button is locked.  
  
@++;  
Gauthier.  
P.S. I'm interested to know more information about the "to_byte" error you encountered and how you overcome it.
I work under Linux Debian, The Windows's test are brief... I'm interested in feedback.

jdeline, 2022-01-24:
Hi Gauthier,

 

I found a solution to the “to_byte” missing method error online.  I had to install a specific version of the python-serial library.

By the way, I recently sent you a 10 Euro donation for grbl-Mega-5X.  I hoped you received it.

 

Jim

fra589, 2022-02-18:
Hi everyone,
  
For the "to_byte" error, it seems to be due to the pyserial module which is outdated. To update it, use the command:
`pip install pyserial --upgrade`
  
@++;  
Gauthier.

closed 2022-02-18 by fra589