fra589/cn5X #36: Windows 10, Think I am doing something wrong. Have followed instructions to the best of my ability. Found 1 other with similar problem. His remedy did not work for me.

survivethewalkingdead, 2022-02-11:


survivethewalkingdead, 2022-02-11:
Microsoft Windows [Version 10.0.19042.1466]
(c) Microsoft Corporation. All rights reserved.

C:\Users\Owner>cd desktop

C:\Users\Owner\Desktop>cd cn5xmaster
The system cannot find the path specified.

C:\Users\Owner\Desktop>dir
 Volume in drive C is Acer
 Volume Serial Number is E8D5-30CC

 Directory of C:\Users\Owner\Desktop

11-Feb-22  09:32 AM    <DIR>          .
11-Feb-22  09:32 AM    <DIR>          ..
06-Apr-15  07:00 PM           546,722 192by86.pdf
18-Mar-16  12:49 PM    <DIR>          3D Print Files
23-Dec-19  08:41 PM         2,396,719 A.STOCK.ctz
30-Jan-22  11:22 PM       117,305,832 arduino-1.8.19-windows.exe
27-Dec-15  01:39 PM        11,378,778 chirp-daily-20151225-installer.exe
30-Jan-22  11:25 PM         1,343,320 ChromeSetup.exe
14-Nov-19  07:50 PM       130,511,968 CitrixWorkspaceApp.exe
11-Feb-22  09:32 AM    <DIR>          cn5X-master
11-Feb-22  09:28 AM         3,110,980 cn5X-master.zip
31-Mar-21  03:28 PM            16,998 Dason 15percenttfsa.ods
15-May-15  12:40 PM    <DIR>          Dason CostaRica
14-Jul-15  09:00 PM    <DIR>          Doris pics 3
05-Dec-20  06:06 PM    <DIR>          DorisPhonePics2020
13-Apr-21  02:25 PM    <DIR>          EFI LOG FILES
02-Jan-20  10:44 PM             2,172 EFILive V7.5 Scan Tool.lnk
02-Jan-20  10:44 PM             2,192 EFILive V7.5 Tune Tool.lnk
26-Feb-19  08:46 AM           388,389 FabCutQuote.xlsx
09-Feb-22  10:08 PM       587,823,472 FreeCAD-0.19.3-WIN-x64-installer-3.exe
31-Jan-22  01:51 AM           308,301 grbl-1.1h.20190825.zip
01-Feb-22  08:25 AM    <DIR>          grbl-Mega-5X-edge
09-Feb-22  10:11 PM           519,237 grbl-Mega-5X-edge.zip
23-Dec-19  08:41 PM         2,378,545 LML_116211_SINGLE_AC.ctz
23-Dec-19  07:57 PM         2,379,263 LML_116211_STKDELETE_AC.ctz
20-Oct-16  10:37 PM           769,585 M146 Spoolmate 150  English.pdf
11-Nov-19  08:45 PM    <DIR>          New folder
14-Oct-15  01:39 PM    <DIR>          NewPingWaterDepth
17-Jul-14  06:27 PM            11,439 Pete dip chart.xlsx
01-Feb-22  07:01 AM        28,239,176 python-3.10.2-amd64.exe
29-Feb-16  05:47 PM    <DIR>          SunhokeyMarlin-Marlin_v1 - 1
14-Oct-15  03:30 PM         2,643,453 SunhokeyMarlin-Marlin_v1.rar
10-Feb-22  09:55 PM             1,404 Visual Studio Code.lnk
02-Dec-15  08:59 AM            51,921 vitodens_warranty.pdf
10-Feb-22  09:53 PM        77,256,912 VSCodeUserSetup-x64-1.64.2.exe
25-Feb-18  02:34 PM        40,221,304 Windows-KB890830-x64-V5.57.exe
              24 File(s)  1,009,608,082 bytes
              12 Dir(s)  310,636,204,032 bytes free

C:\Users\Owner\Desktop>cd cn5x-master

C:\Users\Owner\Desktop\cn5X-master>dir
 Volume in drive C is Acer
 Volume Serial Number is E8D5-30CC

 Directory of C:\Users\Owner\Desktop\cn5X-master

11-Feb-22  09:32 AM    <DIR>          .
11-Feb-22  09:32 AM    <DIR>          ..
11-Feb-22  09:33 AM    <DIR>          cn5X-master
               0 File(s)              0 bytes
               3 Dir(s)  310,636,195,840 bytes free

C:\Users\Owner\Desktop\cn5X-master>cd cn5x-master

C:\Users\Owner\Desktop\cn5X-master\cn5X-master>dir
[... 64 lines cut ...]
               8 Dir(s)  310,636,183,552 bytes free

C:\Users\Owner\Desktop\cn5X-master\cn5X-master>cn5x
cn5X++ v0.8.7 running from: C:\Users\Owner\Desktop\cn5X-master\cn5X-master

                ####### #     #
  ####   #    # #        #   #     #       #
 #    #  ##   # #         # #      #       #
 #       # #  #  #####     #     #####   #####
 #       #  # #       #   # #      #       #
 #    #  #   ## #     #  #   #     #       #
  ####   #    #  #####  #     #

Traceback (most recent call last):
  File "C:\Users\Owner\Desktop\cn5X-master\cn5X-master\cn5X.py", line 2762, in <module>
    window = winMain()
  File "C:\Users\Owner\Desktop\cn5X-master\cn5X-master\cn5X.py", line 430, in __init__
    self.__pBox = qwProgressBox(self)
  File "C:\Users\Owner\Desktop\cn5X-master\cn5X-master\qwprogressbox.py", line 65, in __init__
    self.pBoxLblStart.setGeometry(20, (self.pBoxBtnHeader.height() - self.pBoxLblStart.height())/2, self.pBoxLblStart.width(), self.pBoxLblStart.height())
TypeError: arguments did not match any overloaded call:
  setGeometry(self, QRect): argument 1 has unexpected type 'int'
  setGeometry(self, int, int, int, int): argument 2 has unexpected type 'float'

C:\Users\Owner\Desktop\cn5X-master\cn5X-master>

fra589, 2022-02-12:
Hi @survivethewalkingdead,  
  
Many thanks for reporting,  
It seams to be a bug (it's strange, but it doesn't happen to me) and I think I found where is it. :-)  
I pushed a patch (v0.8.7.20220212), but, since I'm not at home, I don't have a Windows Box to test ans the bug doesn't affect me, tell me if it work for you.  
For traceability, I also need to know what is your version of Windows, Python and QT?  
  
@++;  
Gauthier.

survivethewalkingdead, 2022-02-12:
Thanks for the prompt response! :) 
Edition	Windows 10 Home
Version	20H2
Installed on	‎10-‎Jul-‎21
OS build	19042.1466
Experience	Windows Feature Experience Pack 120.2212.3920.0

Py 3.9*** and 3.10***.Currently 3.10. Oh, I just found 3.4 also installed. Have removed 3.4 and 3.9.
I can't find qt version now...each time I installed successfully a message came up with version and slightly newer version available. I was not able to upgrade it. When I tried pip install pyqt5 just now it did something different, opened a second cmd window for a moment, went back to ready.
I've downloaded the new zip, will go try it now...
Darren.

survivethewalkingdead, 2022-02-12:
Uninstall all python. Download 3.10.2 again. Install. Install as admin. Will not run, will not associate with any .py files. Can't do anything now. :(  Windows is all i know, NOT impressed anymore! (if ever. LOL). Googling...

survivethewalkingdead, 2022-02-12:
re-boot. uninstall py. reboot. install py. reboot. uninstall code reader. Still doesn't associate with Python, but Visual Studio starts it and it looks like it runs! First time I see the control panel! Thank you very much! Will play with it more tomorrow. 
Darren

fra589, 2022-02-13:
Hi @survivethewalkingdead,  
  
I'm please to know that cn5X work for you :smiley:  
Normally, open .py files is a Python install option, this option is not active by default and need to be checked during the Python install procedure.
By the way, perhaps there is a conflict between this option and the Visual Studio default file open options...
  
You can try to change the default file association with one of this link:
https://support.microsoft.com/en-us/windows/change-default-programs-in-windows-10-e5d82cad-17d1-c53b-3505-f10a32e1894d
https://www.google.com/search?q=change+default+file+association+windows
  
@++;  
Gauthier.

closed 2022-02-13 by fra589

fra589, 2022-02-13:
Hi Darren,  
  
AppData is for Data, not for software programs. This probably the reason you can't associate .py files to Python. For this, you have to install it in Programs files.  
  
Yes, pyserial is used for the serial com with Grbl.

The PyQT the [QT5 framework](https://www.qt.io/) in Python. 
The QT5 framework is a cross platform development tool which have many-many functionalities. In cn5X, I use it for driving all the graphical interface windows and widgets, the multithreading, and many other things.  
It works just as well in Linux (which I use) as in Windows or on Mac :-)
  
@++;  
Gauthier.

survivethewalkingdead, 2022-02-13:
Strange python default save/install location. Changed to program files works much better! Thank you! 

Is there documentation anywhere of features, settings, etc? Specifically, how to set up for Plasma cutting, and what are the lower left indicators? x,y,z,a,b,c,p,d,h,r,s? Thanks again,

Darren.

fra589, 2022-02-14:
Hi Darren,  
  
Like all retirees, I don't have enough time to do what I have in mind, and so I haven't found the time to write the full cn5X documentation yet. I rather favor writing new features for cn5X & grbl-Mega-5X...  
I also do a lot of things other than computing software developement :-)  
  
Process specific settings, if any, are to be made at the machine and Grbl level and not in cn5X.  
  
The lower left indicators indicate the state of grbl's inputs:  
x,y,z,a,b,c => Limit switch pins status,  
p,d,h,r,s => The probe, door, hold, soft-reset, and cycle-start pins, respectively.  
  
@++;  
Gauthier.

survivethewalkingdead, 2022-02-14:
I appreciate your efforts and totally understand the "other" distractions etc. :-)  I work 7 days in camp, 7 days off. Can be tough on days-off to squeeze in play time (currently build and learn cnc) in between honey-do tasks. Lol. Last night I read through all the code and comments, got a better understanding of where to make changes. Sorry for asking machine questions in gui section! Right guy, wrong place. :) Just figured that out now. 🤦‍♂️ Now I'll go read Mega5xEdge section on Github.Thanks again,Darren

survivethewalkingdead, 2022-02-15:
Since I can't read this, I don't know if it is Cn5x or Mega5x related... just trying to move steppers at the kitchen table. No limit switches so I removed all Z commands. It could very well be an issue with my gcode, but the random french that pops up may be something you would like to address? If so I can keep sending them to you as I find them. If not, no problem, just let me know! :-)

Oh...just clicked define G28 position and Cn5X closes...

$X
[MSG:Caution: Unlocked]
ok
G17 G20 G90 G94 G54
ok
X-0.5 Y0.
ok
G02 X0. Y0.5 I0.5 J0. F2.5
ALARM:2: Limite logicielle,
Alarme limite logicielle. La cible de mouvement du code G dépasse la course de la machine. Position de la machine conservée. L'alarme peut être déverrouillée en toute sécurité.

fra589, 2022-02-16:
Hi Darren,

It's strange, the default language should be English when the local system language is not known in cn5X. The message you are quoting should have been in English.
I have the same behavior on my own cn5X system, I'll have to find out why this message is in French...

The alarm number 2 you are getting is from grbl-Mega-5X and is normal. This stems from the standard way coordinate space is handled in a CNC milling machine.
By default, the homing procedure (or at power-up when homing is disabled) sets the coordinates to 0 for the maximum axis travel limits. So, according to this, all machine coordinates lie in a space of negative dimensions. If you enter an axis destination with a positive coordinate (Y0.5 in your case) it will generate this error.
To use positive coordinates, the work coordinates must be defined using the GCode commands G92 or/and G54-G59. see http://linuxcnc.org/docs/html/gcode/coordinates.html

About cn5X crash when you click define G28 position, I need more details about the complete procedure which made this crash, from cn5X start to the cn5X unexpected close. So, it will be possible for me to try to reproduce and correct it.

@++;
Gauthier.