fra589/cn5X #37: Crashes when accessing grbl settings

survivethewalkingdead, 2022-02-16:
Found by accident that the grbl settings button was finally not grayed out if estop is pressed.  Pressed, paused, closed. Finally able to catch python output in a video and screenshot. Pic attached.

Is there a graphical process progress view that I'm missing? Or just xyz numbers changing as motors move?

Thank you,
[image]

Darren.

survivethewalkingdead, 2022-02-16:
Had posted in a closed issue, copying relevant parts here...similar problem, no pic...
Just clicked define G28 position and Cn5X closes...

Trying to run a simple file and get this alarm... definitely my fault, moved machine off zero,, set g54, i think, and runs fine...but?

ALARM:2: Limite logicielle,
Alarme limite logicielle. La cible de mouvement du code G dépasse la course de la machine. Position de la machine conservée. L'alarme peut être déverrouillée en toute sécurité.

The random french that pops up may be something you would like to address? If so I can keep sending them to you as I find them. If not, no problem, just let me know! :-)

Darren

survivethewalkingdead, 2022-02-16:
Pic of the g28 crash...

Darren
[image]

fra589, 2022-02-16:
Hi Darren,  

Thanks to open this new issue for new problem.
I found where is this bug :-) I pushed the patch few minutes ago.

This due to your screen resolution which give an odd integer dimension to your main Window. This dimension divided by 2 give a float number which give this error in your system with your QT5 and Python Windows10's specific version. This error doesn't not appear on my Linux system.

This is the same bug than the original problem of #36, now, I made the correction for all dialog boxes!

@++;
Gauthier.
P.S. Don't forget to close this issue if it work for you. If not, tell me more.

survivethewalkingdead, 2022-02-16:
Thank you 😊.  Sorry, I'm on my way to airport for work. I won't be able to test until next Wed.Thanks for the quick work! I'll get back to you as soon as possible.Darren.

survivethewalkingdead, 2022-02-24:
Works fine now! Thanks Gauthier 😊 

Darren.

closed 2022-02-24 by survivethewalkingdead