fra589/cn5X #46: Program aborts immediately upon connect

ccwtruck, 2024-06-24:
Running cn5X-0.8.10_qt5 on Debian 12, attempting to connect to my Mega-5x controller.  I believe that this may be a possible bug... unless my Python3/ QT5 installation is missing a module?

At the bottom of this message is my terminal screen showing all output of the command line "python3 cn5X.py" in user mode (not root)

I have complied with the pre-requisites and have installed everything it called for.  

It is connecting through /dev/ttyACM0 at 115200 baud; the port has been configured for connecting to an Arduino device with the command:
"stty -F /dev/ttyUSB0 cs8 115200 ignbrk -brkint -icrnl -imaxbel -opost -onlcr -isig -icanon -iexten -echo -echoe -echok -echoctl -echoke noflsh -ixon -crtscts " and the Arduino Mega communicates with either the CAT , TAIL or STTY commands:

==================================================
john@shop:~/cn5X-0.8.10_qt5$ tail -f /dev/ttyACM0
0
$22=0
$23=0
$24=25.000
$25=250.000
$26=250
Grbl 1.2h ['$' for help]
^C
===================================================

Error Output On Program Run:
===================================================
QSocketNotifier: Can only be used with threads started with QThread
cn5X++ v0.8.10.20230423 running from: /home/john/cn5X-0.8.10_qt5

                ####### #     #
  ####   #    # #        #   #     #       #
 #    #  ##   # #         # #      #       #
 #       # #  #  #####     #     #####   #####
 #       #  # #       #   # #      #       #
 #    #  #   ## #     #  #   #     #       #
  ####   #    #  #####  #     #

using null output device, none available
Traceback (most recent call last):
  File "/home/john/cn5X-0.8.10_qt5/cn5X.py", line 2524, in on_sig_status
    retour = self.__decode.decodeGrblStatus(data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/cn5X-0.8.10_qt5/grblDecode.py", line 198, in decodeGrblStatus
    self.__mpos[I] = float(tblPos[I])
                     ^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: '0.0000.000'
Aborted
==============================================

fra589, 2024-06-24:
Hi @ccwtruck,  
  
On the cn5X output, I don't know why you have the 2 lines:
> QSocketNotifier: Can only be used with threads started with QThread

and
> using null output device, none available

I don't have those output in my Debian 12.5 box... I don't know why you have it.

But it's not the origin of the main crash!
In your configuration, I actually don't know why, it seems that Grbl's output is not in the expected format.
Perhaps it come from serial communications errors, from language configuration, or many other possibilities.

This kind of error was not managed until now.
In the joined file, I added error management for those errors. When they will occur, the program should not crash, but an error message will appear in the cn5X log tab.
Please, report it for me to understand where the errors come from.

To install this patch, just replace the grblDecode.py file in the cn5X directory and restart cn5X.

@++;
Gauthier. 

[grblDecode.py.zip](https://github.com/user-attachments/files/15957827/grblDecode.py.zip)

ccwtruck, 2024-06-24:
> Hi @ccwtruck,
>
> On the cn5X output, I don't know why you have the 2 lines:
>
>
> and
>
>
> I don't have those output in my Debian 12.5 box... I don't know why you have it.
>
> But it's not the origin of the [...]

And, after replacing the file, here is the output:

`Traceback (most recent call last):
  File "/home/john/cn5X-0.8.10_qt5/grblDecode.py", line 199, in decodeGrblStatus
    self.__mpos[I] = float(tblPos[I])
                     ^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: '0.0000.000'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/john/cn5X-0.8.10_qt5/cn5X.py", line 2524, in on_sig_status
    retour = self.__decode.decodeGrblStatus(data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/cn5X-0.8.10_qt5/grblDecode.py", line 202, in decodeGrblStatus
    self.sig_log.emit(logSeverity.error.value, self.tr("grblDecode.decodeGrblStatus(MPos): ValueError: {}, grblOutput = {}".format(str(e), grblOutput)))
    ^^^^^^^^^^^^
AttributeError: 'grblDecode' object has no attribute 'sig_log'
Aborted
`

fra589, 2024-06-24:
Oh!
I was a little too quick to make the patch!
I will quickly complete it.

But the information made me see that there is a bug in Grbl-Mega-5X with the cloned axes!
The status line should be
<Idle|MPos:0.000,0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000,0.000>
A comma is missing between the coordinates of axes 4 and 5 (Z and 2nd Y)

To work around this, try to uncomment in config.h (line 102):
#define REPORT_VALUE_FOR_AXIS_NAME_ONCE

I will publish a Grbl-Mega-5X patch soon.

@++;
Gauthier

fra589, 2024-06-24:
Do you have made any other changes to grbl-Mega-5X other than axis numbers and names?

I just made a test with a similar config of your:

```
#define N_AXIS 4
#define N_AXIS_LINEAR 4
#define AXIS_1_NAME 'X'
#define AXIS_2_NAME 'Y'
#define AXIS_3_NAME 'Z'
#define AXIS_4_NAME 'Y'
```

And the output of Grbl is good:
```
Grbl 1.2h ['$' for help]
[MSG:'$H'|'$X' to unlock]
$i
[VER:1.2h.20220109:]
[AXS:4:XYZY]
[OPT:VNMGH,35,255,48]
?
<Alarm|MPos:0.000,0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000,0.000>
```

Can you tell me more about your setup ?

@++;
Gauthier.

fra589, 2024-06-24:
And another patch for cn5X to handle this error ithout crash:
[grblDecodeErrorHandling.zip](https://github.com/user-attachments/files/15959463/grblDecodeErrorHandling.zip)

fra589, 2024-06-24:
And a new patch for cn5X v0.8.10_qt5 without crash on error handling:
[cn5X-grblDecodeErrorHandling-v0.8.10_qt5.zip](https://github.com/user-attachments/files/15959522/cn5X-grblDecodeErrorHandling-v0.8.10_qt5.zip)

ccwtruck, 2024-06-24:
> And a new patch for cn5X v0.8.10_qt5 without crash on error handling: [cn5X-grblDecodeErrorHandling-v0.8.10_qt5.zip](https://github.com/user-attachments/files/15959522/cn5X-grblDecodeErrorHandling-v0.8.10_qt5.zip)

Sorry, @fra589 -- no luck; it is still giving me the same error code.

My setup is a 1500x1500mm CNC plasma table.  All 4 axises are 1605 ball screws turned by NEMA 23 motors; drivers are DMA860S drivers from RattMotor.  The Arduino Mega 2560 R3 is wired in the RAMPS1.4 configuration (without the use of a RAMPS board and normally works well.

Oh -- and to answer the question about the second Y being seen as a rotary axis:  It is an OpenBuilds Control issue -- after reading their code, they are empirically assuming that any 4th & 5th axis has to be a rotary, and they are settng the $ grbl settings for those axises for rotary operation.  Their EEPROM $RST=* functions specifically set those axises as rotary.  And the program also assumes that they are rotary, and makes all of their calculations accordingly.  This problem cropped up only after I did a $RST=* in OpenBuilds Control and persisted until I cleared the EEPROM with a sketch.  Now, as long as I don't do a reset in OpenBuilds, it works fine -- and UGCS works fine too.

John

fra589, 2024-06-25:
Hi @ccwtruck,  
  
> I did do a bit of "pruning" in your code; I went through and systematically eliminated axises 5 & 6 (changing #define statements only) so that it now runs 4 LINEAR axises
  
OH NO! It's not the solution for running 4 linear axes with grbl-Mega-5X!!!
There are some other parts in the code which use directly the N_AXIS and N_AXIS_LINEAR definitions in loops and other methods and don't use the `#if N_AXIS == 4` or `#if N_AXIS == 5`...
This can explain your `$RST=*` problem and the problem with the wrong status line which crash cn5X...

For your need: 4 LINEAR axes = X, Y, Z and 2nd Y, you should just put the right parameters in the Grbl's config.h, the compiler will made good job by itself using all the #defines presents in the original grbl-Mega-5X sources and others axes numbers internals calculation.

In grbl-Mega-5X's config.h file (lines 45 to 82), write this:
```
//----------------------------------------------------------------------
// Axis definitions :
//----------------------------------------------------------------------
// IMPORTANT: When changing the axis definitions (axis numbers N_AXIS, 
// linears axis number N_AXIS_LINEAR or axes names AXIS_*_NAME, 
// don't forget to issue the reset factory defaults Grbl command: $RST=*
// if you forget the $RST=* command after change, Grbl may have 
// unpredictable behavior!
//----------------------------------------------------------------------

#define N_AXIS 4        // Number of axes (3 to 6)
#define N_AXIS_LINEAR 4 // Number of linears axis, must be <= N_AXIS

// Axis indexing and names
#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X' // Axis names must be in X, Y, Z, A, B, C, U, V, W, D, E & H.
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'
#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'Y' // Letter of axis number 4
#endif
#if N_AXIS > 4
  #define AXIS_5 4
  #define AXIS_5_NAME 'B' // Letter of axis number 5
#endif
#if N_AXIS > 5
  #define AXIS_6 5
  #define AXIS_6_NAME 'C' // Letter of axis number 6
#endif
#if N_AXIS > 6
  #error "N_AXIS must be <= 6. N_AXIS > 6 is not implemented."
#endif
```

And so that UGS does not believe that there is a 4th rotating axis, still in config.h, uncomment the line number 102:
```
#define REPORT_VALUE_FOR_AXIS_NAME_ONCE
```

Then, cn5X and UGS will work fine, and `$RST=*` too.

@++;
Gauthier.

ccwtruck, 2024-06-25:
> Hi @ccwtruck,
> 
> > I did do a bit of "pruning" in your code; I went through and systematically eliminated axises 5 & 6 (changing #define statements only) so that it now runs 4 LINEAR axises
> 
> OH NO! It's not the solution for running 4 linear axes with grbl-Mega-5X!!! There are some other parts in the code which use directly the N_AXIS and N_AXIS_LINEAR definitions in loops and other methods and don't use the `#if N_AXIS == 4` or `#if N_AXIS == 5`... This can explain your `$RST=*` problem and the problem with the wrong status line which crash cn5X...
> 
> 
> @++; Gauthier.

Yes, @fra589 -- I figured that... after we had the status line trouble, it seemed to work ok in UGCS but not cn5X.  So, I went back and started from scratch with a new copy of Mega-5x.  I configured it as you said (even before you said to  ;-) ) and the trouble went away.  It seems to be running fine in UGCS now, with no "rotation" on the second Y axis, and the status line is correct; I haven't booted into my Debian side of the box yet to test cn5X, but I believe it will be fine.  It does not show any position indicators for the second Y axis (it looks like a normal 3-axis system), but I won't be able to test the cloning until I get my motors re-connected -- I am in the middle of a major upgrade to the table, and I still have to run one more motor cable and my cables for my proximity switch limit switches.  Since Mega-5x supports min/max limit switches, I am using a 24-volt relay board to switch ground for each proximity switch; I am using NO switches, and the relay board gives me the opto-isolation needed for the limit switching system.

I believe that we can deem this issue closed now.  I thank you for your fast answers, and taking the time to work this through.

John

closed 2024-06-25 by ccwtruck